IRNAS/KORUZA-driver-specification #1: CPU - MCU comunication

VojislavM, 2016-06-17:
Koruza driver will be part of the bigger Koruza system, and it will be responsible for driving low-level peripherals. This means there must be some kind of communication between main Koruza CPU and Koruza driver MCU. A majority of systems who have this kind of architecture are using UART communication between two processors. This way they can exchange data and commands. It is necessary to establish a good protocol between two processors, and then it would be more obvious what kind of communication is best.

Picture represents how this is done in [LinkIt Smart 7688 Duo](https://www.seeedstudio.com/item_detail.html?p_id=2574) system

[image]

kostko, 2016-06-19:
I propose a protocol based on [TLVs](https://en.wikipedia.org/wiki/Type-length-value), which makes it extensible in a backwards-compatible way. Each protocol message looks like this:
```
[TLV#1] [TLV#2] ...
```

And each TLV looks like this:
```
[type: 1 byte] [length: 2 bytes] [value: length bytes]
```

All protocol values are in network byte order (big endian). The use of TLVs enables older protocol implementations to skip newer TLVs.

The following TLVs are currently defined:
* Command TLV, containing a 1-byte command identifier.
* Checksum TLV, containing a 4-byte CRC32 checksum of all preceding TLVs.

We need to determine what kind of messages should be exchanged between the CPU and MCU and define the TLVs that we will need in the protocol.

I have prepared a message handling implementation for the above protocol:
* https://github.com/kostko/koruza-driver (example is in `main.c`)

VojislavM, 2016-06-28:
I agree with your suggestion for the protocol. Soon I will send you a list of commands that we need to have, and we can then discuss it further. 

In the mean time, I will implement the code you sent on the STM32 platform for Koruza driver.

VojislavM, 2016-11-03:
All communication between MCU and CPU is done thru UART with TLV protocol.

closed 2016-11-03 by VojislavM