IRNAS/KORUZA-driver-specification #6: IR comunication

VojislavM, 2016-08-04:
MCU should handle low-level IR communication between two Koruza units. Receiving the IR data should be handled in the interrupt routine, something like a parallel process. 
Posible problem could occur with reflection of the IR light, so unit that sends data may receive it back at the same time, which is not good.

kostko, 2016-08-04:
> Posible problem could occur with reflection of the IR light, so unit that sends data may receive it back at the same time, which is not good.

This can be detected by having monotonically increasing sequence numbers.

VojislavM, 2016-11-03:
IR communication is implemented in koruza driver (STM32), need to be tested on longer ranges.
IR link will send 3x8 data + 1x8 CRC and control bits every sec when witi board ask for status.
possible problems:
* high current transmitting diodes - not enough power from power supply
* too small lens for transmitter
* maybe receiver should be more exposed.